platformio/platform-teensy #107: USB serial doesn't work with TEENSY_OPT_SMALLEST_CODE

ktgilliam, 2023-09-05:
Having some trouble using the USB Serial on the teensy, while optimizing for code size.

I've been able to reproduce the issue using Teensy 4.1 and VS Code's platformIO IDE on both Windows 10 and Ubuntu 22.04.

Steps to reproduce:

1. Set up the HelloSerialMonitor Teensy example in VSCode (source copied below).
2. Build and upload, and observe that the serial device is detected by the the OS.
3.  Rebuild with "-D TEENSY_OPT_SMALLEST_CODE" added to build_flags in platformio.ini, then upload and observe that now the serial device does not show up (even if you also have "-D USB_SERIAL" in the build flags as well.

Here is the source :
```#include <Arduino.h>
void setup() 
{                
  Serial.begin(38400);
}
void loop()                     
{
  Serial.println("Hello World");
  delay(1000);
}
```
and here is the platformio.ini:
```
[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
build_flags =
    -D TEENSY_OPT_SMALLEST_CODE
    -D USB_SERIAL
```

Any guidance is appreciated!

PaulStoffregen, 2023-09-06:
Please try with Arduino IDE 2.2.1 and Teensy 0.59.3.

ktgilliam, 2023-09-06:
> Please try with Arduino IDE 2.2.1 and Teensy 0.59.3. After selecting Teensy, use Tools > Optimization to select smallest code.

It seems to work for the test case using the Arduino IDE as you instructed. Are they doing something differently than PlatformIO which I can implement? The actual project I'm working on depends pretty heavily on PlatformIO, so I can't easily port it to the Arduino IDE.

Thanks for your help!

PaulStoffregen, 2023-09-09:
What happens if you try Arduino IDE 2.2.1 and Teensy 1.58.1 (the version PlatformIO is probably using)?

ktgilliam, 2023-09-12:
> What happens if you try Arduino IDE 2.2.1 and Teensy 1.58.1 (the version PlatformIO is probably using)?

Correction to my previous reply... I thought I was using Teensy 0.59.3 but it wasn't actually installed in the Arduino IDE (just selected). Apologies for that. I ran through several tests now after I realized my mistake, and what I'm finding is that with BOTH Teensy 0.59.3 and And 1.58.1, using Arduino 2.2.1 (both installed and from zip), regardless of the optimization setting, the COM port does populate in device manager... for a few moments at a time. It is repeatedly appearing and then disappearing, and if I try to watch the output in Arduino's serial monitor or using PuTTY I'm not seeing anything.

After un/reinstalling Arduino 1.8.19, teensyduino, platformio, making sure all my drivers are up to date and trying all that stuff again, behavior is the same as in the original post.

So it looks like Arduno 2.2.1 is producing whole different set of behaviors.