platformio/platform-teensy #111: Avrdude not available as a valid upload tool

MCUdude, 2023-12-13:
@valeros / @ivankravets Avrdude now supports the AVR Teensy bootloader using `-c teensy`. Even though you may prefer to use the "official" teensy upload tool for PlatformIO, it would still be very useful to have Avrdude available as well.

In my case, I have a custom board that uses the same chip as the Teensy++ 2.0, but I'll have to use Avrdude externally to upload the code using an ISP programmer.

Bottomline: Please add Avrdude (7.2) as an available upload tool 🙂 

```ini
[env]
platform = teensy
board = teensy2pp
framework = arduino

; Environment for ISP upload
[env:MYBOARD_ISP]
upload_protocol = custom
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cusbasp
  -B0.5
upload_command = avrdude $UPLOAD_FLAGS -Ulfuse:w:0x9f:m -Uhfuse:w:0xdf:m -Uefuse:w:0xF3:m -Uflash:w:$SOURCE:i
```

```
Building .pio\build\MYBOARD_ISP\firmware.hex
Configuring upload protocol...
AVAILABLE: custom, teensy-cli, teensy-gui
CURRENT: upload_protocol = custom
Uploading .pio\build\MYBOARD_ISP\firmware.hex
'avrdude' is not recognized as an internal or external command,
operable program or batch file.
*** [upload] Error 1
```

MCUdude, 2024-05-06:
@valeros any plans to look into this?

valeros, 2024-05-06:
Hi @MCUdude, sorry for the late reply. Isn't [platform_packages](https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform_packages.html) option applicable here?  I believe this way you can force PlatformIO to download the package according to your version requirements then just update your custom upload command to include the full path to the Avrdude package, something like this:
```ini
[env:...]
...
platform_packages = 
	tool-avrdude @ ~1.70200.0

upload_protocol = custom
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cusbasp
  -B0.5
  
upload_command = $PROJECT_PACKAGES_DIR/tool-avrdude/avrdude $UPLOAD_FLAGS -Ulfuse:w:0x9f:m -Uhfuse:w:0xdf:m -Uefuse:w:0xF3:m -Uflash:w:$SOURCE:i
```