platformio/platform-teensy #112: Teensy 1.59 has been released

skunktrading, 2024-02-09:
https://forum.pjrc.com/index.php?threads/teensyduino-1-59-released.74436/

PaulStoffregen, 2024-02-09:
Please be aware Teensy is now compiling as C++17.  See #109.

closed 2024-02-09 by valeros

skunktrading, 2024-02-10:
Thanks for updateing so quickly :)
I was able to switch to 1.59 by putting
`platform = https://github.com/platformio/platform-teensy.git`
into my platformio.ini
In order to load the matching tools package I had to add
`platform_packages =
    tool-teensy @ 1.159.0
`
into my platformio.ini

Is this expected? I seem to remember that previous updates handled switching to the matching tools etc. without needing the override.

PaulStoffregen, 2024-02-10:
Is PlatformIO using the libraries we publish for Teensy to use with Arduino IDE?

For the 1.59 release, they are all in this tarball.

https://www.pjrc.com/teensy/td_159/teensy-package-1.59.0.tar.zst

Asking because we recently had a lengthy forum thread where someone had all sorts of problems with mismatched SD and SdFat libraries... and eventually admitted they were using PlatformIO rather than Arduino IDE.

PaulStoffregen, 2024-02-10:
Especially with SD cards, the core library, SD (unique for Teensy) and SdFat (fork for Teensy) we publish for Teensy really need to be used together.

When users install some other copy of SD or SdFat, things go badly.  :(

skunktrading, 2024-02-10:
Hi Paul
As far as I can tell the correct libraries are definitely being used

PaulStoffregen, 2024-02-10:
Thanks.  That's quite a relief to hear.

commit f8e6babe (2024-02-09): Update Teensyduino to v1.59

Resolves #112, resolves #109