platformio/platform-teensy #115: Native ethernet broken between 4.17 and 4.18

ernestum, 2024-03-01:
After long hours I found out that native ethernet breaks between the above mentioned releases. Probably it is an upstream issue, probably it is related to #108

I use the following code to test it.

```c++
#include <Arduino.h>
#include <SPI.h>
#include <NativeEthernet.h>

// Initialize the Ethernet server library
// with the IP address and port you want to use
// (port 80 is default for HTTP):
EthernetServer server(80);

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Ethernet WebServer Example");

  // start the Ethernet connection and the server:
  uint8_t mac[6];
    for (uint8_t by = 0; by < 2; by++) mac[by] = (HW_OCOTP_MAC1 >> ((1 - by) * 8)) & 0xFF;
    for (uint8_t by = 0; by < 4; by++) mac[by + 2] = (HW_OCOTP_MAC0 >> ((3 - by) * 8)) & 0xFF;
  Ethernet.begin(mac);

  // Check for Ethernet hardware present
  if (Ethernet.hardwareStatus() == EthernetNoHardware) {
    Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
    while (true) {
      delay(1); // do nothing, no point running without Ethernet hardware
    }
  }
  if (Ethernet.linkStatus() == LinkOFF) {
    Serial.println("Ethernet cable is not connected.");
  }

MDNS.begin("myteensy", 1);
MDNS.addService("_http._tcp", 80);

  // start the server
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());
}

void loop() {
  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
          client.println("<!DOCTYPE HTML>");
          client.println("<html>");
          // output the value of each analog input pin
          for (int analogChannel = 0; analogChannel < 6; analogChannel++) {
            int sensorReading = analogRead(analogChannel);
            client.print("analog input ");
            client.print(analogChannel);
            client.print(" is ");
            client.print(sensorReading);
            client.println("<br />");
          }
          client.println("</html>");
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```

With the ethernet shield attached to the teensy and my computer in the same network I test the ethernet connection using

```bash
wget -O - myteensy.local
```

With anything above v4.17 it works the first time (if at all) and the second time wget fails to connect.

valeros, 2024-03-04:
Hi @ernestum, does it work as expected in Arduino IDE?

PaulStoffregen, 2024-03-05:
People have reported NativeEthernet is broken in Arduino IDE with the latest versions (since we updated to the gcc 11 toolchain).

QNEthernet is the preferred library now.

valeros, 2024-03-05:
Closing as not related to PlatformIO. As a workaround just downgrade your platform version to the last known working one, for example:
```ini
[env:teensy41]
platform = teensy @ ~4.17.0
framework = arduino
board = teensy41
```

closed 2024-03-05 by valeros

ernestum, 2024-03-05:
Thanks valuable info! Would have been valuable to have this in the release notes @PaulStoffregen

PaulStoffregen, 2024-03-05:
Nobody new (or at least I didn't know) when the release notes were written.

Very likely we'll delete NativeEthernet and FNET from the next release, and bundle QNEthernet instead.  That would certainly be in the next version's release notes.