platformio/platform-teensy #118: Possibility of using newer ARM GCC toolchain or even an LLVM toolchain?

spnda, 2025-05-10:
To clarify, I don't know how this toolchain stuff is setup or how the teensy toolchain is decided on exactly. However, over on the ARM GNU Toolchain [download site](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) there are much newer versions of the `arm-gnu-eabi` gcc toolchain available than what this platform package currently offers. The `toolchain-gccarmnoneeabi-teensy` package currently ships GCC 11.3.1 iirc, but as new as 14.2 is available on the arm page. I briefly tested my own project which seems to compile and work fine. The only new warnings/errors that come up is from invalid/bad code GCC 11.3.1 didn't catch yet. I don't really see a reason not to upgrade, since it brings a lot of fixes, improvements, library features, and especially new warnings/analyzers that could be very helpful to embedded developers.

I also want to use this issue to ask about the possibility of using LLVM for Teensy development. LLVM and Clang are cross-compilers by nature, and essentially only require a special toolchain (so runtime libraries, headers, ..., not the actual compilers) for each target configuration. I'm not entirely sure on the details, and I could imagine the PlatformIO core libraries aren't flexible enough to support this. There's plenty of reasons to use Clang over GCC in my opinion, so I'd love to see support coming to the Teensy package. If other people aren't interested or don't have time to do this, I'd be happy to get some pointers in the right direction for how I could potentially add a compatible LLVM toolchain for Teensy.

PaulStoffregen, 2025-05-11:
Historically PlatformIO has used the toolchain PJRC publishes for use with Teensy on Arduino IDE.  Pretty unlikely that will change.

To discuss future PJRC software direction, post on this forum:

https://forum.pjrc.com/index.php?forums/suggestions-bug-reports.4/

Just to set your expectation, PJRC has historically updated toolchain versions about once every 5 to 6 years.  Every time has involved a lot of work, usually because the lesser used platforms aren't provided as compiled binaries and need a full toolchain build.

Each time some libraries or special programs unexpectedly broke.  For example, going from 5.4 to 11.3 seemed to work, but after the first release with 11.3 we discovered subtle C++ static initialization order problems because C++ doesn't technically guarantee const init of C++ classes with constexpr constructors, even though gcc 5.4 always had that behavior.  A C++ language purist might argue we were wrong to depend on undefined C++ behavior.  But the practical reality is we're still on C++17 which doesn't offer any way to guarantee such behavior.  Extreem purists would probably then argue that nobody should build programs to depend on such behavior.  But with Teensy we aim to support this and many other things that make use of the very common C++ Arduino API classes simpler for the reality of the way novice programmers build their applications.  Turned out gcc 11.3 does only give the const init behavior we needed in certain situations.  Some library classes, especially HardwareSerial, Wire, SPI were updated to restore this not-guaranteed-by-compiler behavior to work with gcc 11.3.

My point here is we have a history to infrequent toolchain updates because of these painful problems.  This is absolutely not a decision to be made lightly.  We will not update simply because a newer version is available.  We also never consider changing language dialect (eg, go from C++17 to C++20) within 2 releases of a toolchain update!

There must be tangible and compelling reason to update.  The simple fact that a newer version exists is not a compelling reason.

closed 2025-05-12 by valeros

spnda, 2025-05-12:
> But the practical reality is we're still on C++17 which doesn't offer any way to guarantee such behavior.

Well, with the current use of static initialization, no. However, there's ways around it like for example the [construct on first use](https://isocpp.org/wiki/faq/ctors#static-init-order-on-first-use) idiom. I have used this quite a lot in other projects outside of embedded and given that the initialization of static variables within functions is defined and thread-safe, this would be the best option.

I wouldn't call myself a C++ purist by any means, but it feels like a cheap excuse to say updating is not an option because it will break our current code because we've got bugs we don't want to fix. It's undefined behavior, and it always was, who knows what could happen. However, I do understand that that is because of bad design choices with Arduino.h and is now technical debt that can't be fixed without breaking code (or at least I don't see any other way myself). So it feels like a cheap excuse but seems to be the harsh reality, and I am an offender myself with my Teensy code as it turns out. Perhaps something like zephyr would not have these issues? Haven't gotten around to testing it out but if it's better designed it shouldn't really have these issues.

I always try to consciously write "correct" code that avoids undefined behavior as much as possible because of things like this, and I like to live on the cutting edge. But thank you for giving the insight, and I hope I can help in some way in the future to make updates less painful. I will try and test my code as much as possible using GCC 14.2 locally, and see if I can find any issues, though so far so good with two non-trivial projects.

> To discuss future PJRC software direction, post on this forum:
>
> https://forum.pjrc.com/index.php?forums/suggestions-bug-reports.4/

I've tried posting on the forums before but my post never got approved and then deleted at some point. I think I tried contacting you before over email about this.